VanceVagell/kv4p-ht #224: [Android] Randomly transmits without input

Hotwire, 2025-02-05:
Good day,

v1.8c board, brand new freshly built. Latest firmware.

It worked for about 20 min, then started to key up and TX on its own (as evidenced by the blue LED, and the KV4P-HT App indicating via the PTT turning pink/red).

Any suggestions

VA3HQB

VanceVagell, 2025-02-05:
Is the setting "beacon my position" turned on?

Also, are chatting with someone on APRS when this happens?

Hotwire, 2025-02-05:
No, neither. I have done nothing with APRS.

Hotwire, 2025-02-05:
So, it may be a board. I have 4, 3 of them work just fine, its just the one that seems to get flakey after about 10 min, it seems to heat up around the main board shield. NOT the ESP32 shield.

VA3HQB

dkaukov, 2025-02-05:
@Hotwire you can try to pull-up GPIO5 and  GPIO33 to 3.3v with 1k resistors.

VanceVagell, 2025-08-02:
Closing this out, having not enough information to act on this. If others can reproduce this, please reopen.