VanceVagell/kv4p-ht #232: Is the radio module SA818S supported?

lohriialo, 2025-02-12:
[Quick start guide](https://kv4p.com/quick_start.html) says this about the radio module : 
`SA818-V radio module (DRA818V may be substituted)`

Couldn't find the exact module, please advise if the following module can be a replacement for PCB version 1.7b
- https://hubtronics.in/docs/SA818S.pdf
- https://hubtronics.in/sa818s-uhf-walkie-talkie-module

VanceVagell, 2025-02-12:
> Yes, SA818S will work fine!

Just be sure to get the VHF version. The UHF version will work (and the app
supports it), but you'd need to replace the VHF filter components on the
PCB with UHF filter components.

closed 2025-03-09 by VanceVagell

TMMXA, 2025-04-10:
@VanceVagell I would like to ask if I need to change the PCB or just modify the parameters of the original components if I use the SA818S-U module

And how to replace the VHF filter components on the PCB with UHF filter components.