VanceVagell/kv4p-ht #234: [Android] Ensure physical PTT button logic can't bypass tx restriction outside ham band

VanceVagell, 2025-02-19:
From Bill K7WFB:

" I think there may be a bug in app version 1.7.3 when used with the version 2d board: When tuned to a frequency out of the ham bands (e.g. Wx on 162.4250), if you press a physical PTT button, the app indicates that you are transmitting (on-screen PTT turns red). It stays in that condition until the app is closed. I don't believe it is actually transmitting, however. Additional info: I pressed the right-hand PTT button. Phone A - Pixel 7a/Android 15; Phone B - OnePlus 6/Oxygen 11."

Bill4965, 2025-03-04:
On retesting, I get different results. Here's the latest.

Test conditions: 

- app version 1.7.3 
- board 2d 
- Phone - Pixel 7a/Android 15
- I pressed the right-hand PTT button

Test protocol:

1. Tune to a frequency out of the ham bands (e.g. Wx on 162.4250)
2. Press a physical PTT button.
3. Note that the app indicates that you are transmitting (on-screen PTT turns red). In fact, you are not transmitting.
4. Tune to a ham band frequency.
5. Note that the on-screen PTT is still red.  In fact, you are not transmitting.
6. Press a physical PTT button to transmit, then release it.
7. Note that the on-screen PTT is now white.

N.B. I could not reproduce this bug on my other phone: OnePlus 6/Oxygen 11.

VanceVagell, 2025-03-29:
I think this might have been caused by bug #239 which will be fixed in release 1.7.4.

closed 2025-03-29 by VanceVagell