VanceVagell/kv4p-ht #237: [Android] Add more ways to tune/change frequency

norrinr, 2025-02-22:
Please add more ways to change the frequency beyond typing in the number. Maybe a slider bar/wheel or + / - with a step?

raff, 2025-05-01:
Also, I though "scan" would scan the band, not just cycle through the memorized frequencies (and I couldn't understand why "scan" wasn't doing anything until I added a couple of channel :)

Would be good to have a "scan band" command, that stops when it finds a signal.

VanceVagell, 2026-03-03:
@raff we're tracking the request to scan frequencies in #94 so keep an eye on that for future developments!

In terms of adding buttons to move up and down by small amounts, this is not currently planned for the project, since the use case for it is limited and a goal of the project is to focus on features most people need so the radio remains more approachable. The reality for most hams is that they either want to tune to a known simplex frequency to talk to buddies (walkie talkie style) or they want to program in nearby repeaters (or use our "Find nearby repeaters") function to tune to specific repeater frequencies.

There are certainly cases where moving up/down a little bit might be helpful, but it's niche enough that we wouldn't change the main UI to accomodate it. Thanks for the idea though!

closed 2026-03-03 by VanceVagell