VanceVagell/kv4p-ht #238: Potential new features that would interest paragliders.

92ajmills, 2025-02-24:
So I'm an absolute novice with all this stuff but I do have some ideas for the device and the app. I'm a paraglider and there isn't a great option to use a bluetooth headset (eg Shokz openrun pro) with a radio and a Bluetooth PTT button. For sub 2500ft the app called Zello works well but above that ham radios work best. 

My question is could you make the app either work with Zello, or allow headphones and a Bluetooth PTT whilst allowing the app to sit in the background? It will be amazing if the radio app could work in the background and activated by a bluetooth PTT button. 

Paragliders use navigation apps like seeyounavigator or XCtrack while zello works in the background. 

If something like this could be done it would dominate the coms market for paragliders, hang hangliders, and any other motorless aviators FYI. I have some UK paraglider distributer contacts that would jump on this if it worked well! Happy to help with if needed. 

I've bought the device and look forward to trying it out. Many thanks

mikechristiansen, 2025-04-18:
I second this! Also a huge market for Paramotoring. 
I wonder how this would work if the bluetooth headset was paired with the phone, would the audio from the radio come through? A bluetooth button or a hardwired button could then be used for PTT and the app would hopefully run in the background. Im looking to order the new 2.x board with UHF for testing of these functions.

vitosaver, 2025-10-26:
I would love this! The whole paragliding community would love this, because there is no budget radio on the market that has bluetooth. This would make it easy to connect bluetooth PTT button and headset like Sena or Cardo.