VanceVagell/kv4p-ht #246: Can you Help with These ESP-32 Modules

tpd33, 2025-03-10:
Hi,

I recently bought these at Aliexpress, the problem is that I can't find manuals, datasheet, symbols and footprints for KiCAD

[image]
[image]

.
Does anyone have them?

Thanks

tpd33, 2025-03-10:
[image]

petrkr, 2025-04-25:
You do not need that.

Just put there two 1x30 pin headers and pinout you have labeled on those boards.. Datasheet is just same as for every ESP32.

That second one is ESP32-S3, which does not require USB-UART controller.. Would be nice if v2 board use it instead of out-of-dated ESP32-WROOM-D (-E is still active)