VanceVagell/kv4p-ht #248: [Android, ESP32] Significant audio quality issues on both rx and tx since recent changes

VanceVagell, 2025-03-15:
Behavior on HEAD:
- APRS rx and tx fail (cannot decode incoming APRS messages, and the messages it sends cannot be decoded by another device, I tested with kenwood TH-d74)
- Rx audio has a repetitive popping sound

I tested this on a kv4p HT PCB v2.0c.

These issues do not exist when I roll back to commit 99939fc29d64ab1112d909527cb6725ad51c33a7 (the last one that's in v1.7.3 release), and re-test with the same devices. Audio is fine, and APRS works fine in both directions.

This is a significant regression that will block the next release until we fix them. @dkaukov would you prefer I revert to that working commit and cherry-pick other new commits to give you time to fix this, or do you think you could get this working in the next week or so? This must be related to your recent re-work.

dkaukov, 2025-03-16:
I’ll take a look and update soon.

VanceVagell, 2025-03-16:
> @VanceVagell <https://github.com/VanceVagell> ,
>
> I’ll take a look and update soon. Can you confirm that you're using the
> latest main branch versions for both the firmware and the app?

Yes, confirmed.

VanceVagell, 2025-03-16:
I also just tested with PCB v1.7x and the same issue exists.

However, after several more tests, I've found that rx audio on the kv4p HT is the primary issue. It often has repeating pops/skips, and it cannot decode APRS messages. Reverting to the commit noted above fixes the issue.

Several tests sending APRS text to a nearby Kenwood TH-D74A worked OK. So I'm not sure if tx audio is a big issue (it's at least intermittent as an issue), but APRS rx is definitely broken.

dkaukov, 2025-03-16:
Hi @VanceVagell,

Please see the linked PR  https://github.com/VanceVagell/kv4p-ht/pull/249. I was able to reproduce and fix the APRS issues; however, I did not encounter the audio problems you've reported. My audio functions correctly, but currently, I don't have access to a board with the standard CP2102 to test under those conditions. If you could test on your end and provide feedback, it would be greatly appreciated. Should the audio issues persist, I can assemble test boards, though this would require an evening or two to complete.

VanceVagell, 2025-03-22:
Unfortunately, PR #249 did not resolve this. I merged it optimistically and retested at head.

See attached video:
https://github.com/user-attachments/assets/95b1ed9e-6001-4d5f-8193-e7be133cea7f

The first half of the video is a test with a v1.7b PCB, the second half of the video is with a v2.0c PCB. I'm just holding down PTT on another HT (Kenwood TH-D74A) and you can that the audio has some kind of pulsing distortion. Once in a while an APRS message will decode (you can see a message decode at the end of the video), but the vast majority I tested could not be decoded. 

Tx audio is fine, outgoing APRS messages seem consistent.

It sounds like perhaps there are frequent buffer underruns?

dkaukov, 2025-03-23:
Hi @VanceVagell 

Sounds like RSSI queries—these can be audible when RX is overloaded. Can you try:

Setting the RSSI period to 500ms

Reducing TX power (I use attenuators for this)

Let me know if that helps!

dkaukov, 2025-03-23:
Hi @VanceVagell,  

I recreated your experiment and got similar results.  

With a 40dB attenuator on TX:  
🔗 https://www.youtube.com/shorts/5rpE91drHrQ  

Without it:  
🔗 https://www.youtube.com/shorts/mDFg2lJParQ  

The clicking noise in the second video sounds a lot like what you're experiencing. What’s confusing me is that, in v1.7.3, I can also hear RSSI queries in this condition—clicking twice per second.

VanceVagell, 2025-03-29:
Thanks for the additional tests. I tested 1.7.3 and 1.7.2 and find that this is intermittently a problem at least that far back. Sometimes there's no clicking/skipping at all, othertimes there is. Closing this bug out for now since it's inaccurate as currently described, but we should keep an eye on this and see if we can reproduce it more readily so we can fix it.

closed 2025-03-29 by VanceVagell

VanceVagell, 2025-04-13:
I went back to 1.6.x releases before the audio changes and still hear this. So I'm not sure what's going on with it.