VanceVagell/kv4p-ht #254: [Android, ESP32] First use of physical PTT sometimes gets stuck on (intermittent)

VanceVagell, 2025-04-13:
This only happens on a v2.0 PCB with physical PTT button(s). To repro (intermittent):

1. Plug in kv4p HT device
2. App opens automatically on phone
3. Press and release the physical PTT Button

Expected result:
4. Device stops tx.

Actual result:
4. Device is stuck in tx mode.

The software PTT button (circle on bottom of app screen) shows red when tx is stuck on. User can tap the software ptt button to get out of the stuck tx, and then it seems the physical PTT button works fine after this.

It very likely has something to do with the new debounce utilities in utils.h / buttons.h in the firmware source. However, since I can't reproduce this consistently, it's not easy to find a fix for it.

dkaukov, 2025-04-13:
Hi @VanceVagell, would it be possible to grab some logs when this happens? Also, what happens if you press and release the physical PTT button again — does that change anything?

dkaukov, 2025-04-25:
Hey @VanceVagell — did you have any chance to test out https://github.com/VanceVagell/kv4p-ht/pull/256 ? Curious how it’s looking on your end!

closed 2025-04-27 by VanceVagell