VanceVagell/kv4p-ht #263: [ESP32] Some pins may introduce instability if left floating

VanceVagell, 2025-04-24:
See discussion from Discord:
https://discord.com/channels/1290074006908829746/1290074125553111143/1364829935688089672

The short version is that at least GPIO4 causes audio issues when left floating, and it may just be good to pull down all the unused pins on the ESP32.

dkaukov, 2025-04-25:
This kind of sounds like a red herring.

If the issue is happening in RX mode, GPIO interference is usually pretty minimal. And since GPIO4 isn’t even on ADC1, it shouldn’t be affecting anything there.

It might be worth checking for a hardware fault on that board — maybe a bad ground or flaky solder joint — rather than focusing on unused pins.

dkaukov, 2025-04-26:
Seems like hardware issue: https://discord.com/channels/1290074006908829746/1290074125553111143/1365484096654147676
(The SQ routing does not match selected HW version)

VanceVagell, 2025-04-27:
Closing this out since it seems like @dkaukov found another way to fix this. Please let me know if this issues remains, and feel free to reopen this if needed.

closed 2025-04-27 by VanceVagell