VanceVagell/kv4p-ht #278: #278 Sketch proof of concept to detect board features, SQ Pin, ADCBias present, Physical PTT and H/L capability

BillH63, 2025-05-07:
[KV4P_Hardware_Detection.txt](https://github.com/user-attachments/files/20076630/KV4P_Hardware_Detection.txt) 

Sketch does the following:

- Determines active GPIO pin for Squelch 4 or 32
- Determines if the ADCBias circuit is present
- Determines if physical PTTs are present

With the above information it can infer the board version and hence the presence of the H/L circuitry
The table shows the conditions required to determine version and H/L (helps with issue #267 ).

[image]

The code does also pull the hardware strapped version which can be used to determine 2.0c boards for attenuation level.

It might be possible to tweak the ADCBias code to detect the difference between 2.0c and 2.0d but I don't have a 2.0c to test. If debug is enabled the value of adcDiff will be displayed and it can be seen if it is different enough between versions to set a min and max threshold for each version. Otherwise, the hardware strapped version can be used for 2.0c detection.

Detection of the Physical PTT can be used to bypass PTT code if they are not present and might help resolve issue #224 

In the end if the features present can be separated from the hardware strapped version it would allow someone to for example mod a 1.7b board to add Physical PTTs by simply adding the circuitry to the appropriate pins. The 10K pull up resistor is detectable in code.

dkaukov, 2025-05-08:
Hi @BillH63, Interesting approach! I might be missing part of the picture, but I think the main issue we’ve been circling is how HW revisions currently bundle both pin mappings and feature availability. So if someone wants to enable just one feature (like H/L power), they end up needing to “pretend” it’s v2.0c and also deal with unrelated quirks.

That said, if features like the H/L switch can be reliably auto-detected, that would definitely move us in the right direction. Appreciate the sketch — it’s close, though I think it’s solving a slightly different problem than the one we’ve been discussing.

BillH63, 2025-05-08:
Hi @dkaukov, for now the presence of H/L can only be inferred by the board version. In order to make it detectable something would have to be done like require GPIO23 be tied to GPIO22 when H/L is added to a board that isn't known to support it. Then it could be seen if one follows the other. The other features I added are all detectable in their own way.

dkaukov, 2025-05-08:
@BillH63 Maybe just adding a pull-down on GPIO22 would do the trick? Makes detection easier, and it’s also better hardware-wise — right now the FET gate is floating since GPIO22 is in high-Z.

BillH63, 2025-05-08:
I thought of that too, but haven't tested it. After fighting with GPIO5 for the Right PTT and trying to detect it I didn't want to propose it until tested. FYI GPIO5 when not connected will read HIGH no matter what the input mode is set to. If I connected it to ground it did go low, but was HIGH otherwise. Tested on several of the dev boards I have and several of the HT revisions and it was always HIGH. So, changed the code to only look at the Left PTT which did behaved.

VanceVagell, 2025-08-02:
Due to the recent firmware updates to use non-volatile storage (NVS) so that kit builders can pre-configure the ESP32 to match the provided radio module, this is less useful going forward. But thank you for looking into it, @BillH63 !