VanceVagell/kv4p-ht #282: Proposal: Use NVS for Centralized Board Hardware Configuration

dkaukov, 2025-05-08:
We're proposing to use NVS (`nvs_flash`) to store board-specific hardware configuration (`hwconfig`), such as pin mappings and enabled features. This would allow us to support multiple hardware variants with a single firmware binary, without cluttering the codebase with conditionals and quirks.

📄 **Comparison Matrix:**
| Approach                                    | Pros                                                                                                                                                                                                                                                                                                                                                                             | Cons                                                                                                                                                                                                                        |
| ------------------------------------------- | -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- | --------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------- |
| **Centralized `hwconfig` in NVS**           | - Single firmware binary for all boards<br>- Clean and structured config<br>- “Everything enabled” config covers most cases<br>- Scales easily as new boards appear<br>- NVS only needed for non-standard designs<br>- Breaks the dependency between hardware and firmware release cycles — tweaks don't need new firmware<br>- @Smitty is happy to preflash 2.x boards with NVS | - Slightly more setup for custom boards<br>- Recovery from corrupted/missing NVS could confuse users<br>- Requires some knowledge/tools to manage NVS                                                                       |
| **GPIO-based detection + scattered tweaks** | - Also a single firmware binary<br>- Works automatically for known designs<br>- Doesn’t require users to understand NVS — just connect resistors to strap pins according to table                                                                                                                                                                                                | - Firmware gets cluttered with board-specific quirks<br>- Harder to scale cleanly with new board variants<br>- Requires firmware updates for every new layout |

**Benefits:**

* Clean separation of firmware and hardware config
* One firmware binary for all boards
* Easier testing: “everything enabled” config works nearly everywhere
* Simplifies support for future boards without firmware changes
* No need to wait for a firmware release just to support a new layout

**Considerations:**

* NVS is only needed for non-standard/custom boards
* Adds a bit of complexity for people deviating from the standard layout
* Recovery from corrupted/missing NVS needs to be understandable
* Some concerns about DIY usability and tooling (Android app, online flasher?)

SmittyHalibut, 2025-05-09:
I 100% agree this is the right solution going forward.

I disagree with a few details provided in the summary:

* There will be just as many conditionals throughout the code.  If a feature is conditional, it'll be conditional.  The only difference is where the state for that conditional is determined.  But I don't think this is a big deal; no solution would be any better.
* I don't think we should bank on "NVS only needed for non-standard configs" and just accept that NVS is always there, and do some sane things when it's not.  I can foresee forks of the hardware where one fork goes one way and the other fork goes the other way.  Two standards, doing things differently.  NVS would be needed in that case to differentiate, and that's fine.

I'll add to this design:
* If NVS is not present, then the firmware and app should work together with the user to determine the correct values, then program them into NVS.  If there are good defaults, then present those defaults to the user, letting them over-ride things as necessary.
  * This is a good place to put the "try to determine features by detecting hardware" code.
* Even if NFS *IS* present, I think there should be a mechanism for the user to over-ride it and save new values.  With big warning signs: "This could break functionality if you get this wrong. Only do this if you know what you're doing, and WHY you're doing it."  I'm a Unix user; give the users all the rope they need to sink the ship, but at least warn them.
  * There should also be a "Reset to defaults" button for when users invariably get it wrong. 'cuz Users gunna Use.

But, in general, I VERY MUCH like an arbitrary key/value store to configure individual features, instead of trying to derive features from a single "hardware version" indicator (like I had designed earlier.  I like this design much better.)

dkaukov, 2025-05-26:
Hi @SmittyHalibut,

Got it — we're definitely on the same page overall. Just wanted to clarify a few things:

1. **Conditionals** – All the feature toggles are centralized in `board.h` and evaluated at runtime during startup. Most of the time it’s just about skipping the relevant `xxxSetup()` and `xxxLoop()` calls. That’s been working well so far, though yeah, we might run into trickier cases down the line.

2. **Defaults** – I’m not relying on NVS being used only for non-standard setups. It just makes sense to have solid defaults baked in — I chose PCB 1.x as the baseline. NVS can override that if needed, but we’re not blocked without it.

3. **Flashing NVS** – Totally agree the app should be able to flash NVS. I just don’t think it’s worth reverse-engineering the partition format. Easier path: store pre-built NVS binaries on GitHub and let the app download and flash the right one. (We could push NVS writing into the firmware, but feels like a lot of work for not much gain.)

Big yes to letting users override stuff (with clear warnings), and definitely a "reset to defaults" option. And yeah — key/value config feels way more flexible than trying to force everything through a single hardware version ID.

dkaukov, 2025-06-15:
Hi @SmittyHalibut, where would you like the v2.0c and v2.0d board definitions to go? Looks like we don’t have a standard folder structure for them yet. ;)

closed 2025-07-13 by VanceVagell