VanceVagell/kv4p-ht #285: kv4p module does not scan and log states: Backtrace: 0x40083de9:0x3ffbeb9c |<-CORRUPTED

GreenBeeSolutions, 2025-05-13:
kv4p module does not scan and log states: Backtrace: 0x40083de9:0x3ffbeb9c |<-CORRUPTED
When plugged into the phone no error shows up, I get a noise but when starting the scan nothing is found

As the module does not work I installed it via the webbrowser to get the log

Any idea what's wrong?

[16:59:15]ޭ��ޭ��Dޭ��UE (20260) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[16:59:15]E (20260) task_wdt:  - loopTask (CPU 1)
[16:59:15]E (20260) task_wdt: Tasks currently running:
[16:59:15]E (20260) task_wdt: CPU 0: IDLE0
[16:59:15]E (20260) task_wdt: CPU 1: loopTask
[16:59:16]E (20260) task_wdt: Aborting.
[16:59:16]
[16:59:16]abort() was called at PC 0x40104e7c on core 0
[16:59:16]
[16:59:16]
[16:59:16]Backtrace: 0x40083de9:0x3ffbeb9c |<-CORRUPTED
[16:59:16]
[16:59:16]
[16:59:16]
[16:59:16]
[16:59:16]ELF file SHA256: eed1f79226991170
[16:59:16]
[16:59:16]Rebooting...
[16:59:16]ets Jul 29 2019 12:21:46
[16:59:16]
[16:59:16]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[16:59:16]configsip: 0, SPIWP:0xee
[16:59:16]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[16:59:16]mode:DIO, clock div:2
[16:59:16]load:0x3fff0030,len:1184
[16:59:16]load:0x40078000,len:13232
[16:59:16]load:0x40080400,len:3028
[16:59:16]entry 0x400805e4

dkaukov, 2025-05-16:
Hi @GreenBeeSolutions, thanks for the report!

Looks like there might be two things going on:

1. **Scan not finding anything** — Just to clarify, the scan in kv4p only checks **channels**, not raw frequencies, and only works if **squelch is on**. If squelch is off (and you hear noise), or no channels are set, it won’t show anything — that’s normal.

2. **Installed via browser** — Sounds like flashing might have gone wrong. Give it another try.

dkaukov, 2025-06-15:
Hi @GreenBeeSolutions , any news? ;)