VanceVagell/kv4p-ht #297: APRSdroid / Kiss TNC

sysadminpower2019, 2025-06-20:
It would be great to enable the app to pipe audio / data to aprsdroid to use it as a KISS TNC

Conversly it would be great to also allow the hardware itself to communicate via bluetooth to aprsdroid independently if possible

dkaukov, 2025-07-03:
Hi @sysadminpower2019,  I took a quick look at APRSdroid, and from what I can see, kv4p already implements all the core functionality. What specific use-case are we aiming to cover by integrating with APRSdroid? Just trying to understand what’s missing or what we’d gain from the integration.

sysadminpower2019, 2025-07-10:
Hey thanks for getting back.

Theres actually a lot of reasons

1) it's more then aprsdroid, Woad, Xastir etc all use the same interface

2) KISS TNC is an standard that been in use for a rather long time in ham
radio. That means that by providing an interface via KISS you are actually
improving compatibility and adoption of an existing standard

3) This also means you can focus your time and energy on further firmware
improvements which are device specific instead of trying to do everything
at once. It saves you time and increases usability for all hams and co

4)Sharing with the community means not only honoring and propagating and
exiting standard but also simplifying contributions to your project and
others that are related.

5) it's just a nice thing to do and helps all of us out.

If you need any more info or reasons why this is a benefit for us all
please let me know. My aim is simply to make it easier for everyone to play
along and save time

dkaukov, 2025-07-10:
Hi @sysadminpower2019 ,  got it — really appreciate you laying out the rationale.

Just to clarify: in kv4p, all AFSK encoding/decoding is done on the Android side, not on the hardware module. That’s an intentional choice — the phone has way more processing power and memory, which lets us keep the hardware simple and cost-effective.

Because of that, implementing a TNC interface would only really make sense *within the app*, likely over TCP. There’s no serial-accessible TNC on the hardware itself. And at that point, you’re essentially running a software TNC alongside an app that already handles APRS natively.

So the question is: what exactly would the TNC interface enable in this setup, given that kv4p already implements the APRS functionality directly? Just trying to understand the added value in practical terms.

sysadminpower2019, 2025-07-10:
I see I understand.

None the less afaik LORA-APRS uses similar ESP32 and is able to implement a
HW KISS TNC

https://github.com/richonguzman/LoRa_APRS_Tracker

That being said TCP is probably also viable for some applications

APRSdroid,direwolf and Xastir afaik do accept TCP streams directly.

.

dkaukov, 2025-07-11:
Thanks — yeah, I’ve seen LORA-APRS. The key thing there is that encoding and decoding is offloaded to the LoRa hardware itself, which is purpose-built for that kind of packet handling. Very different from kv4p, where we need to digitise analag audio and run software decoder. 

Also, if everything’s running on the same device, linking apps together over TCP to duplicate functionality the kv4p app already provides feels kind of pointless. Unless there’s a real need to plug into external systems, it just adds complexity for no gain.

sysadminpower2019, 2025-07-11:
Ah i understand

Well the app can't also do winlink for example and the other advanced
options from aprsdroid, xastir, woad etc.

I'd say it would be an opportunity to focus on enhancing the firmware and
extracting as much from the hardware while allowing the current app to stay
relevant but less focused.

At the end we have many independent, non connected proprietary apps. Here
there is the possibility to improve the hardware / firmware to make it
compatible with the broad existing world of ham software.

What we have less and is harder to do is good hardware / firmware here is
where this project can really shine without having to reinvent the world as
well

> <
https://ci3.googleusercontent.com/meips/ADKq_NbLop825xQeLH4RlhMDLHBWx9AbRgmK-ZZg6C65CG0x2hikBPh6XiQXElN9MJ29bhxjqLqhcs-yFkUglWbh_hMfageSOQM6wi530k-rCg=s0-d-e1-ft#https://avatars.githubusercontent.com/u/4616540?s=20&v=4>dkaukov
left a comment (VanceVagell/kv4p-ht#297)
>
> Thanks — yeah, I’ve seen LORA-APRS. The key thing there is that encoding
and decoding is offloaded to the LoRa hardware itself, which is
purpose-built for that kind of packet handling. Very different from kv4p,
where we need to digitise analag audio and run software decoder.
>
> Also, if everything’s running on the same device, linking apps together
over TCP to duplicate functionality the kv4p app already provides feels
kind of pointless. Unless there’s a real need to plug into external
systems, it just adds complexity for no gain.
https://ci3.googleusercontent.com/meips/ADKq_NY4yDhNJboEQJBzfQ4UlrNbGZ5XIOW1F66Zm_qCXPSvrlh2XvklTAQ9Cv0VWUN9O9FWA4_Smj8tWp-BmDSEpqjZFzcR6_RsWLn69sDwKUe5EU3yIA-EP_1GrFFe7kLruYyTUnpitDaa7e05MSSIiuut0WwSOItwMiJNB1WTDMiuat5TQYeR_3kKkNPoL5wv7pkfmAwuZeQ4IjTdllPRwipvk16IeBlZvW-WCeErAg7qktEXCtsaSiM=s0-d-e1-ft#https://github.com/notifications/beacon/AJWXABQS536GRJ7EKXHYART3H4E77A5CNFSM6AAAAAB7YNSFWGWGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTVWMFMTS.gif>Message
ID: ***@***.***>

dkaukov, 2025-07-11:
Hi @sysadminpower2019, thanks again for the input — it seemed like an interesting idea at first, but without clear use cases it’s starting to feel like a bit of a dead end. `kv4p` is meant to be a simple app + hardware setup, with all the APRS handled on the Android side. It works well for what it’s meant to do.

If a solid reason to add a TNC interface shows up later, I’m open to it — but for now, I don’t think it’s worth the extra effort.

VanceVagell, 2025-07-12:
Note that I would like to have the android app offer a KISS TNC interface
over Bluetooth. That shouldn't be too hard (famous last words) and
leverages all the existing logic on the app for tuning, filters, etc.
without us needing to maintain lots of different clients. But could open up
some fun use cases where you have your phone and a laptop on the go, and
want to do some packet radio on the laptop.

sysadminpower2019, 2025-07-13:
thanks for your reply.
I think that would be an acceptable and great compromise solution.
Would be happy to beta test it and support this option.

dkaukov, 2025-07-15:
Hi @na7q  — thanks for the note. I’m guessing that by “KV4P device” you mean the little box with the antenna on one end and USB-C on the other.

Let me also take a guess at your use case: you want to plug the KV4P device into your phone, use the KV4P app to talk with friends over voice, then close it and open APRSdroid to send a few messages using the same device. Is that more or less what you're thinking?

If so, that’s probably not going to work — we can’t really multiplex TNC, audio, and rig control over a single serial connection in a reliable way. But if you had something else in mind, definitely let me know!

na7q, 2025-07-23:
> Hi [@na7q](https://github.com/na7q) — thanks for the note. I’m guessing that by “KV4P device” you mean the little box with the antenna on one end and USB-C on the other.
> 
> Let me take a guess at your use case: you want to plug the device into your phone, use the KV4P app to talk with friends over voice, then close it and open APRSdroid to send a few messages — all using the same hardware and firmware, without needing to flip any kind of "mode switch." Is that about right?
> 
> If so, that’s probably not going to work — we can’t reliably multiplex TNC, audio, and rig control over a single USB serial connection (and the AFSK modem isn’t in the box — though we can ignore that for now). Also, I’m assuming you're not suggesting changes to APRSdroid or other apps, since this would rely on using a standard KISS TNC over USB serial.
> 
> If you had something else in mind, I’m definitely open to hearing it!

I don't care to talk on it at all. I mean, it could be useful, but idea is entirely APRS. Just having the ability to pipe the audio into APRSDroid would be very useful for doing packet (minus the AFSK issues on newer devices). I'm sure hardware changes to the original KV4P device could allow for these options, or even a built in TNC using a simple micro arduino or other design. All just ideas to bring the tech to the next level.

dkaukov, 2025-07-24:
Hi @na7q, thanks for sharing the use case — that helps clarify things. For what you’re describing, https://github.com/nakhonthai/ESP32APRS_Audio looks like a better fit. It runs a full AFSK modem directly on the ESP32 and can act as a standalone TNC terminal, which means it doesn’t need the kv4p app or external audio handling. Should work well with the `kv4p` style hardware.

sysadminpower2019, 2025-07-24:
If i understand correctly this sorta goes with the idea of taking advantage
of the hardware to turn it into a powerful digital modes solution for 144m.
So I think your partial solution of the app doing the TNC part is a good
way to start

nerdalertdk, 2025-08-28:
Yes please add support for KISS TNC
This would be a big step for cheap kiss tnc support to ATAK

sysadminpower2019, 2025-12-27:
Hi @VanceVagell just wanted to ask if you have had any time to think about this?