VanceVagell/kv4p-ht #299: NeoPixel glitches on 2.0c board when powered from laptop

dkaukov, 2025-07-08:
**What we found**:

* The NeoPixel data line has a 100 Ω resistor and a 2.2 nF cap (C12) to ground — meant as a low-pass filter.
* That filter cuts off around 723 kHz, which is too close to the NeoPixel’s 800 kHz signal.
* On top of that, ESP32 sends 3.3 V data, which is borderline.
* When plugged into a laptop, the USB voltage is a bit higher (5.1 V), which raises the NeoPixel’s threshold for "HIGH" — and now the signal sometimes doesn’t register.

**Fix**:
Removed C12 → NeoPixel works fine again even from the laptop.

SmittyHalibut, 2025-07-08:
Agreed.  100pF is a good choice.  This was on 2.0c?  I think I found and already fixed this on v2.1 actually; I ended up using a different neopixel chip that was much less tolerant of this problem.

SmittyHalibut, 2025-07-30:
Removed the filter entirely.  It's really not needed.

v2.0e PR incoming.  It mentions this ticket, so it should be linked soon. (I haven't submitted it yet, so I don't have a PR number.)

Resolving.

closed 2025-07-30 by SmittyHalibut